harlequin-tech/WiFlyHQ #44: Unable To Receive Multicast Packets/Join Multicast Group

om37, 2015-05-05:
I am unable to receive UDP packets sent on the default multicast address (224.0.0.1).
Using the sendto() command - though I receive the Err 2FewArgs error - I am able to send UDP packets to the multicast address and have them received by a Java based client. 
The problem comes when receiving multicast packets on the Arduino/WiFly module.

I have looked through the provided doxygen and found no reference - maybe I am missing something - but I think there needs to be some way of joining the multicast group.

For clarity, my Arduino code:

    #include <SoftwareSerial.h>
    SoftwareSerial wifiSerial(8,9);//RX, TX    
    #include <WiFlyHQ.h>
    
    WiFly wifly;
    
    const char   mySSID[]         = "SSID";
    const char   myPassword[]     = "Pass";
    
    void setup() 
    {
        char buf[64];
        Serial.begin(115200);    
        Serial.println("Starting sketch Proc UDP");
    	
        wifiSerial.begin(9600);
        if (!wifly.begin(&wifiSerial, &Serial))
        {
            Serial.println("Failed to start wifly");
        }
        
        Serial.println("Joining network");
        wifly.setSSID(mySSID);
        wifly.setPassphrase(myPassword);
        wifly.enableDHCP();
        
        if (wifly.join())
        {
            Serial.println("Joined wifi network");
        }
        else
        {
            Serial.println("Failed to join wifi network");
        }    
        
        Serial.println("WiFly ready");
    
        Serial.print("IP: ");
        Serial.println(wifly.getIP(buf, sizeof(buf)));
        Serial.print("Netmask: ");
        Serial.println(wifly.getNetmask(buf, sizeof(buf)));
        Serial.print("Gateway: ");
        Serial.println(wifly.getGateway(buf, sizeof(buf)));
    
        Serial.println("Set DeviceID");
        wifly.setDeviceID("Wifly-UDP-Rec");
        Serial.print("DeviceID: ");
        Serial.println(wifly.getDeviceID(buf, sizeof(buf)));
    
        wifly.setIpProtocol(WIFLY_PROTOCOL_UDP);
    }   
        
    int count = 0;
    void loop()
    {
        if(wifly.available() > 0)
    	{
    		int EoL = 0;
    		String s;
    		while(wifly.available() > 0) //EoL == 0)//while(wifly.available() > 0)
    		{
    		  if(wifly.available() > 0)
    		  {
    			  char rec = wifly.read();
    			  if(rec != '\n')
    				s += rec;
    			  else
    				EoL = 1;
        		  }
    		}
    		EoL = 0;  
    		Serial.println(s);
    	}
        Serial.println(wifly.available());    
    }
    
UDP packets sent on the local subnet broadcast (in my instance, 192.168.2.255) are received and displayed as exptected.
When sent to 224.0.0.1, however, wifly.available() only returns 0.

Other clients are able to receive the packets when sent on Multicast, so I know that they are being sent.

harlequin-tech, 2015-11-21:
I can't find any reference to multicast support in the WiFly's data sheet or user guide.  It's possible it does not support it.  I'll try some experiments.

om37, 2015-11-23:
I can't remember where I found clarification but the chips, it turns out, do not support multicast. This issue can now be closed.
Many thanks for the reply after this time.

closed 2015-11-23 by om37