harlequin-tech/WiFlyHQ #45: Tosduino Fails

quique123, 2015-11-20:
Has anyone tried this library with a Tosduino UNO R3?

Im working on it because it seems to fail to make the http request/post sometimes.  Im still trying to track down why it fails to open a connection to a site.

harlequin-tech, 2015-11-21:
The Tosduino looks fully compatible with the Arduino Uno R3.  I didn't see any areas for concern from looking at the board.  I did see this quote on http://tinyosshop.com: "Tosduino UNO R3 board is our new product! It's 100% compatible with original Arduino, now it has perfect quality and  performance , we believe it will provide exciting use experience to you!"  which makes me wonder if some of the boards had production problems.

closed 2015-11-21 by harlequin-tech

quique123, 2015-11-21:
I have a sketch which works perfectly with the Uno but fails with the Tosduino.  Specifically it fails to post data to the Web application.  It opens the connection but posting sometimes works and sometimes fails.  I haven't researched it enough to figure why it fails.

MV