harlequin-tech/WiFlyHQ #54: debug statements

chrwei, 2016-04-13:
there seems to be several different kinds of debugging prints, in including some redundancies like 

#ifdef DEBUG
  DPRINT(whatever);
#endif

and

if (debugon) {
  DPRINT(whatever);
}

if DEBUG isn't set the DPRINT gets optimized out anyway, no need for any form of if.

I have optimized it in my branch, but a pull won't go unless the formatting pull is merged first.

my changes save 1650 bytes of flash with debug off, and 70 bytes with debug on.  the savings with debug on are due to removing the "if (debugon)" versions.

harlequin-tech, 2016-04-13:
The debugon boolean allows you to enable debugging just when wanted.  Useful for debugging specific commands, features, or problems with an app.  There may be a way to improve it with just a direct test in the DPRINT macro.

chrwei, 2016-04-13:
that was not implemented consistently.  there is a dbgBegin() that does allow dumping some trace info no matter of debugon or not, I kept that intact.

adding debugon to the macro would allow turning debug on and off as needed, as long as DEBUG is already defined.  I can see that being useful.

closed 2016-08-22 by harlequin-tech