harlequin-tech/WiFlyHQ #56: Cannot compile example httpserver.ino

Sterpa, 2016-04-24:
I can not compile the example file httpserver.ino
Gives an error message:
...WiFlyHQ-master/WiFlyHQ.h:78:28: error: section attribute not allowed for 'prog_char'
 typedef const char PROGMEM prog_char;

What could be the problem?

duhovny, 2016-04-25:
Hey @Sterpa, I had same problem with the last version of Arduino IDE. Rolling back to v1.0 helped to solve this issue.

Also @chrwei has mentioned he fixed this for last version of Arduino IDE in his forked repository. I haven't tried though.

chrwei, 2016-04-25:
that line does not cause problems for me in 1.6.5.  I'm not sure what the issue is in this case.

Sterpa, 2016-04-25:
Very strange... i use 1.6.8 and @chrwei fofk, and cannot solve this... only remark this lines with #define PROGMEM 

```
//#undef PROGMEM 
//#define PROGMEM __attribute__(( section(".progmem.data") )) 
#undef PSTR 
#define PSTR(s) (__extension__({static prog_char __c[] PROGMEM = (s); &__c[0];})) 
```

but i am not shure, that it is right way. Of course rolling back to v1.0 is a wrong way, we go forward, not downshift))

chrwei, 2016-04-25:
so... I followed the posted error instead of actually open the example, and removing those lines should fix it, or maybe wrap them in an ifdef so older versions still use it?

see pull request #57

harlequin-tech, 2016-08-22:
Update merged, thanks.  This code was an old work-around to avoid compiler warnings when using PROGMEM.  Newer Arduino releases have fixed the problem.

closed 2016-08-22 by harlequin-tech