harlequin-tech/WiFlyHQ #64: Error when try to connect

iFernandoSousa, 2016-11-17:
I'm trying to use the WiFlyHQ to connect to my RN-171 and Xbee Shield.

If I open console and send $$$ I received the message CMD and can get/set all commands.
But when I try any sample of WiFlyHQ I get the message bellow:

> Starting
> Free memory: 1051
> setPrompt failed
> Failed to enter command mode
> Failed to start wifly

I already have changed my SSID and Pass, and set correct RX,TX port in the samples.

Have another thing to do?

My RN: https://www.sparkfun.com/products/11047
My Xbee Shield: http://www.lightinthebox.com/pt/xbee-arduino-v3-0-modulo-compativel-escudo_p381442.html

JoranJix, 2017-05-02:
same problem here. Did you find a solution?

iFernandoSousa, 2020-06-08:
@JoranJix  I change the device :/

closed 2020-06-08 by iFernandoSousa